epsilonrt/mbpoll #1: No Make File For Raspberry Pi

mypiandrew, 2016-09-13:
Hi

This is a great project!

I am super impressed :-)

I want to compile this on a raspberry pi to use ttyS0 UART but I cannot compile it as your installation instructions for RPi appear incomplete.

I do not understand why you should need qmake installed to compile this, can you provide a more basic make file that would work without installing this first?

Thanks for helping 

Andrew.

epsilonrt, 2016-09-13:
Hi Andrew,
Thank you.

mbpoll is a multi-platform project. I chose to use qmake even if I do not use the QT library because I know him well (unlike cmake).
If you want to compile on raspberry pi (like debian), you must install qmake.
Here's the procedure:

    $ sudo apt-get install build-essential qt4-make
    $ cd mbpoll
    $ qmake
    $ make
    $ sudo make install

This should work. If that does not work, it is necessary that you indicate to me what phase it stops and the error message.

Small note: a raspberry pi there is no ttyS0, the serial port is ttyAMA0 (or ttyUSB0 ... if you use a USB-serial adapter)

Best regards

epsilonrt, 2016-09-13:
There is an error on the name of the qmake package in the message below, the correct name is qt4-**qmake**. I just checked on a raspberry pi with Raspbian Jessie recently installed and I have not had a problem.
Regards

mypiandrew, 2016-09-13:
Hi

OK, after installing qmake

I get this error

```
root@raspberrypi:~/modpoll/mbpoll-master# ls
3rdparty  INSTALL  LICENSE     mbpoll.project    NEWS     README_FR.md  src   version-git.h
AUTHORS   LICENCE  mbpoll.pro  mbpoll.workspace  package  README.md     TODO
root@raspberrypi:~/modpoll/mbpoll-master/src# qmake
qmake: could not find a Qt installation of ''
root@raspberrypi:~/modpoll/mbpoll-master# cd src/
root@raspberrypi:~/modpoll/mbpoll-master/src# pwd
/root/modpoll/mbpoll-master/src
root@raspberrypi:~/modpoll/mbpoll-master/src# qmake
qmake: could not find a Qt installation of ''
```
Any help?

mypiandrew, 2016-09-13:
ps. Yes you can activate ttyS0 using device-tree overlays.  I'm running on a custom Compute-Module based board so I have a proper RS232-RS485 card :)

mypiandrew, 2016-09-13:
Hi

OK, found the solution here :

https://github.com/EliasOenal/multimon-ng/issues/43#issuecomment-190222502

```
root@raspberrypi:~/modpoll/mbpoll-master# ls -la $(which qmake)
lrwxrwxrwx 1 root root 9 Sep 25  2014 /usr/bin/qmake -> qtchooser
root@raspberrypi:~/modpoll/mbpoll-master# find /usr -name qmake | grep qt4
/usr/lib/arm-linux-gnueabihf/qt4/bin/qmake
/usr/share/qt4/bin/qmake

root@raspberrypi:~/modpoll/mbpoll-master# alias qmake="/usr/lib/arm-linux-gnueabihf/qt4/bin/qmake"
root@raspberrypi:~/modpoll/mbpoll-master# qmake
root@raspberrypi:~/modpoll/mbpoll-master# ls
3rdparty  INSTALL  LICENSE   mbpoll.pro      mbpoll.workspace  obj      README_FR.md  src   version-git.h
AUTHORS   LICENCE  Makefile  mbpoll.project  NEWS              package  README.md     TODO
root@raspberrypi:~/modpoll/mbpoll-master# make
gcc -c -pipe -Wno-unused-parameter -Wno-unused-function -O2 -Wall -W -D_REENTRANT -D_TTY_POSIX_ -DQT_NO_DEBUG -I/usr/share/qt4/mksp                           ecs/linux-g++ -I. -I/usr/include/qt4 -I3rdparty -I3rdparty/modbus -I. -o obj/mbpoll.o src/mbpoll.c
gcc -c -pipe -Wno-unused-parameter -Wno-unused-function -O2 -Wall -W -D_REENTRANT -D_TTY_POSIX_ -DQT_NO_DEBUG -I/usr/share/qt4/mkspecs/linux-g++ -I. -I/usr/include/qt4 -I3rdparty -I3rdparty/modbus -I. -o obj/modbus.o 3rdparty/modbus/modbus.c
gcc -c -pipe -Wno-unused-parameter -Wno-unused-function -O2 -Wall -W -D_REENTRANT -D_TTY_POSIX_ -DQT_NO_DEBUG -I/usr/share/qt4/mkspecs/linux-g++ -I. -I/usr/include/qt4 -I3rdparty -I3rdparty/modbus -I. -o obj/modbus-data.o 3rdparty/modbus/modbus-data.c
gcc -c -pipe -Wno-unused-parameter -Wno-unused-function -O2 -Wall -W -D_REENTRANT -D_TTY_POSIX_ -DQT_NO_DEBUG -I/usr/share/qt4/mkspecs/linux-g++ -I. -I/usr/include/qt4 -I3rdparty -I3rdparty/modbus -I. -o obj/modbus-tcp.o 3rdparty/modbus/modbus-tcp.c
gcc -c -pipe -Wno-unused-parameter -Wno-unused-function -O2 -Wall -W -D_REENTRANT -D_TTY_POSIX_ -DQT_NO_DEBUG -I/usr/share/qt4/mkspecs/linux-g++ -I. -I/usr/include/qt4 -I3rdparty -I3rdparty/modbus -I. -o obj/modbus-rtu.o 3rdparty/modbus/modbus-rtu.c
gcc -c -pipe -Wno-unused-parameter -Wno-unused-function -O2 -Wall -W -D_REENTRANT -D_TTY_POSIX_ -DQT_NO_DEBUG -I/usr/share/qt4/mkspecs/linux-g++ -I. -I/usr/include/qt4 -I3rdparty -I3rdparty/modbus -I. -o obj/delay.o 3rdparty/sysio/delay.c
gcc -c -pipe -Wno-unused-parameter -Wno-unused-function -O2 -Wall -W -D_REENTRANT -D_TTY_POSIX_ -DQT_NO_DEBUG -I/usr/share/qt4/mkspecs/linux-g++ -I. -I/usr/include/qt4 -I3rdparty -I3rdparty/modbus -I. -o obj/serial.o 3rdparty/sysio/serial.c
g++ -Wl,-O1 -o mbpoll obj/mbpoll.o obj/modbus.o obj/modbus-data.o obj/modbus-tcp.o obj/modbus-rtu.o obj/delay.o obj/serial.o    -L/usr/lib/arm-linux-gnueabihf -lpthread
root@raspberrypi:~/modpoll/mbpoll-master# make install
install -m 755 -p "mbpoll" "/usr/local/bin/mbpoll"
strip "/usr/local/bin/mbpoll"
root@raspberrypi:~/modpoll/mbpoll-master# mbpoll
mbpoll: device or host parameter missing ! Try -h for help.
root@raspberrypi:~/modpoll/mbpoll-master# mbpoll -h
usage : mbpoll [ options ] device|host [ writevalues... ] [ options ]

ModBus Master Simulator. It allows to read and write in ModBus slave registers
                         connected by serial (RTU only) or TCP.

Arguments :
  device        Serial port when using ModBus RTU protocol
                  COM1, COM2 ...              on Windows
                  /dev/ttyS0, /dev/ttyS1 ...  on Linux
                  /dev/ser1, /dev/ser2 ...    on QNX
  host          Host name or dotted IP address when using ModBus/TCP protocol
  writevalues   List of values to be written.
                If none specified (default) mbpoll reads data.
                If negative numbers are provided, it will precede the list of
                data to be written by two dashes ('--'). for example :
                mbpoll -t4:int /dev/ttyUSB0 -- 123 -1568 8974 -12
General options :
  -m #          mode (rtu or tcp, TCP is default)
  -a #          Slave address (1-255 for rtu, 0-255 for tcp, 1 is default)
                for reading, it is possible to give an address list
                separated by commas or colons, for example :
                -a 32,33,34,36:40 read [32,33,34,36,37,38,39,40]
  -r #          Start reference (1 is default)
  -c #          Number of values to read (1-125, 1 is default)
  -u            Read the description of the type, the current status, and other
                information specific to a remote device (RTU only)
  -t 0          Discrete output (coil) data type (binary 0 or 1)
  -t 1          Discrete input data type (binary 0 or 1)
  -t 3          16-bit input register data type
  -t 3:hex      16-bit input register data type with hex display
  -t 3:int      32-bit integer data type in input register table
  -t 3:float    32-bit float data type in input register table
  -t 4          16-bit output (holding) register data type (default)
  -t 4:hex      16-bit output (holding) register data type with hex display
  -t 4:int      32-bit integer data type in output (holding) register table
  -t 4:float    32-bit float data type in output (holding) register table
  -0            First reference is 0 (PDU addressing) instead 1
  -1            Poll only once only, otherwise every poll rate interval
  -l #          Poll rate in ms, ( > 100, 1000 is default)
  -o #          Time-out in seconds (0.01 - 10.00, 1.00 s is default)
Options for ModBus / TCP :
  -p #          TCP port number (502 is default)
Options for ModBus RTU :
[... cut ...]
  -d #          Databits (7 or 8, 8 for RTU)
  -s #          Stopbits (1 or 2, 1 is default)
  -P #          Parity (none, even, odd, even is default)
  -4            RS-485 mode (/RTS on after sending)
  -5            RS-485 mode (/RTS on when sending)

  -h            Print this help summary page
  -V            Print version and exit
  -v            Verbose mode.  Causes mbpoll to print debugging messages about
                its progress.  This is helpful in debugging connection...
root@raspberrypi:~/modpoll/mbpoll-master#
```

mypiandrew, 2016-09-13:
Hopes this help you, thanks for a great project!

closed 2016-09-13 by mypiandrew

mypiandrew, 2016-09-13:
Lovely!!!

root@raspberrypi:~# mbpoll -a 1 -b 9600 -t 4 -r 1 -c 10  /dev/ttyS0 -m rtu
mbpoll 0.1-23 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright (c) 2015 epsilonRT, All rights reserved.
This software is governed by the CeCILL license <http://www.cecill.info>

Protocol configuration: Modbus RTU
Slave configuration...: address = [1]
                        start reference = 1, count = 10
Communication.........: /dev/ttyS0,       9600-8E1
                        t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register, output (holding) register table

-- Polling slave 1... Ctrl-C to stop)
[1]:    12961
[2]:    13072
[3]:    13183
[4]:    12850
[5]:    12850
[6]:    12850
[7]:    12850
[8]:    12850
[9]:    12850
[10]:   12850
-- Polling slave 1... Ctrl-C to stop)
[1]:    12961
[2]:    13072
[3]:    13183
[4]:    12850
[5]:    12850
[6]:    12850
[7]:    12850
[8]:    12850
[9]:    12850
[10]:   12850
-- Polling slave 1... Ctrl-C to stop)
[1]:    15531
[2]:    15642
[3]:    15753
[4]:    15420
[5]:    15420
[6]:    15420
[7]:    15420
[8]:    15420
[9]:    15420
[10]:   15420
^C--- /dev/ttyS0 poll statistics ---
3 frames transmitted, 3 received, 0 errors, 0.0% frame loss

everything was closed.
Have a nice day !

byrrt, 2017-01-09:
Hi, can u send me executables files for raspberry pi? I am also looking for modpoll for raspberry pi. Thx!

epsilonrt, 2017-01-11:
I have added simple compilation instructions on the project homepage. Best Regards

byrrt, 2017-01-11:
Ok, I worked it, sorry for the problem and thank you :)
In the evening, I see it is works witch RTU module.

byrrt, 2017-01-11:
Yes, everything works - again warm thank you!

richi-chen, 2017-08-23:
Hi epsilonrt,

I want to build mbpoll in OpenWRT system running on MTK MT7688 which is MIPS CPU.
How do you think it should be built to an OpenWRT package for mbpoll?

Thanks!

epsilonrt, 2017-08-23:
Hi, 
I thank you to open another issue because your question has nothing to do with this thread.
Regards