epsilonrt/mbpoll #2: endian swap feature

mypiandrew, 2016-09-13:
Can I suggest you add the feature to swap the byte order for 32 bit reads (either long ints or floats)

This comes up more often than you'd expect in the field :-(

christianhals, 2016-10-20:
I second this request. That would be an awesome feature. 

Maybe a new parameter option where you can switch both bitorder and word order. For instance for 32bit you could have ABCD as the default for MSB-LSB AND MSW-LSW and then do CDAB for MSB-LSB and LSW-MSW and so on.

epsilonrt, 2016-10-22:
Hi,
I just created the temporary branch 'big-endian' and add the -B option to enable big endian word order (for reading and writing the entire 32-bit and float).
I have no slave 'big-endian': Could you try and tell me if this change is right for you ?
If you confirm, I merge changes in branch 'Develop'.

mypiandrew, 2016-10-23:
Hi

You can use this program to simulate a modbus RTU/TCP slave

https://sourceforge.net/projects/modrssim2

You can then create all the register "types" by entering the raw values in and reading them back

epsilonrt, 2016-10-23:
Hi,
I have very little time to devote to this project and you are at the origin of this request. Could you tell me if the realized version matches your request. If it does not, I would delete the branch 'big-endian' because I do not need. I would like a minimum of participation. Thank you.

christianhals, 2016-10-23:
Amazing **epsilonrt**. Thanks for the new update. I have physical slaves available at the office so I can verify tomorrow.

christianhals, 2016-10-24:
I'm happy to confirm that the new endian swap feature is working as expected for me.

**Testresult LSW-MSW (Carlo Gavazzi EM24):**

```
# mbpoll /dev/ttyM0 -m rtu -b 9600 -P none -a 36 -c 01 -r 00 -t 3:int -0 -1 -o 2.0 -v
mbpoll 0.1-23 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright (c) 2015 epsilonRT, All rights reserved.
This software is governed by the CeCILL license <http://www.cecill.info>

Opening /dev/ttyM0 at 9600 bauds (N, 8, 1)
Protocol configuration: Modbus RTU
Slave configuration...: address = [36]
                        start reference = 0, count = 1
Communication.........: /dev/ttyM0,       9600-8N1
                        t/o 2.00 s, poll rate 1000 ms
Data type.............: 32-bit integer (little endian), input register table

-- Polling slave 36...
[24][04][00][00][00][02][76][FE]
Waiting for a confirmation...
<24><04><04><09><44><00><00><CC><CF>
[0]:    2372
```

**Testresult MSW-LSW (Circutor CVM-1D):**

```
mbpoll-big-endian# mbpoll /dev/ttyM0 -m rtu -b 9600 -P none -a 03 -c 1 -r 00 -B -t 3:int -0 -1 -o 2.0 -v
mbpoll 0.1-23 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright (c) 2015 epsilonRT, All rights reserved.
This software is governed by the CeCILL license <http://www.cecill.info>

Opening /dev/ttyM0 at 9600 bauds (N, 8, 1)
Protocol configuration: Modbus RTU
Slave configuration...: address = [3]
                        start reference = 0, count = 1
Communication.........: /dev/ttyM0,       9600-8N1
                        t/o 2.00 s, poll rate 1000 ms
Data type.............: 32-bit integer (big endian), input register table

-- Polling slave 3...
[03][04][00][00][00][02][70][29]
Waiting for a confirmation...
<03><04><04><00><00><09><3D><1F><C5>
[0]:    2365
```

Thanks again:-)

epsilonrt, 2016-10-25:
Fine ! Thanks. I merge the big-endian branch in develop branch, this improvement will be integrated into the master branch when I will spread the next version. Thank you for your participation.

closed 2016-10-25 by epsilonrt

mypiandrew, 2016-10-25:
Try this program to help you debug 

https://sourceforge.net/projects/modrssim2