epsilonrt/mbpoll #4: Creating a debian package

epsilonrt, 2016-10-25:
As soon as I have a moment, I'll have done a debian package for mbpoll ...

mmalyj, 2017-11-11:
Did you ever create a debian package for mbpoll? I ask because I can follow your instructions if I'm using an Ubuntu Server, since I can do `sudo apt-get install cmake build-essential` there. But when I'm using Ubuntu Core there is only Snappy and no apt-get, and there is no snap for installing build-essential. That's why a debian package would come in useful. I will then try turning it into a snap with https://larry-price.com/blog/2016/12/07/creating-a-snap-from-existing-deb-packages/.

epsilonrt, 2017-11-11:
Thanks, yes mbpoll has been transferred under cmake and according to the documentation on the github project page, creating a debian package is now possible.

closed 2017-11-11 by epsilonrt

mmalyj, 2017-11-14:
Ah, that worked. It created mbpoll-1.3-Linux.deb, which I was able to install via dpkg in a different instance of Ubuntu, without having to install cmake or build-essential. I'll let you know if I'm successful at turning it into a snap.