epsilonrt/mbpoll #6: Register Addressing Bug

woodrowbarlow, 2017-07-26:
When I execute the following command, my server responds with the value of the **first** holding register (which is correct).

    mbpoll <ip_address> -B -t 4:hex -r 1 -c1 -1

when I execute the following command, my server responds with the value of the **second** holding register (which is incorrect).

    mbpoll <ip_address> -B -t 4:int -r 1 -c1 -1

Unless I'm mistaken, the request sent by mbpoll should be identical in both of these cases... the part after the colon should only affect how mbpoll parses the response, right?

epsilonrt, 2017-07-29:
Hi,
You have effectively misunderstood the use of the suffix **:int**

As indicated in the help page, the suffix **:int** reads a **32-bit integer**, so the **-c1** option indicates that you want to read a **32-bit register**, so two 16-bit registers! As you provided a **-B** option the 2nd register read is the low-order one. The **-B** option is unnecessary for reading a 16-bit register because the order of the bits is imposed by the MODBUS standard.

By default, mbpoll displays integers in decimal, so if you want to read a 16-bit register and display it in decimal you have to do:

`mbpoll <ip_address> -t4 -r1 -c1 -1`

If you want to display this same register in hexadecimal, you must do:

`mbpoll <ip_address> -t4:hex -r1 -c1 -1`

woodrowbarlow, 2017-07-31:
I see. Thank you!

closed 2017-07-31 by woodrowbarlow