epsilonrt/mbpoll #8: mbpoll to simulate a Modbus slave?

mmalyj, 2017-11-15:
I would like to use mbpoll in place of [ModbusPal](http://modbuspal.sourceforge.net/) as a Modbus slave simulator for testing purposes. Is that possible? ModbusPal is GUI based, but my lightweight Ubuntu 16 Core OS does not allow GUI, which is why I like mbpoll since it runs from the command line. But when I try to simulate slave 1 writing value 42 to register 2 :

`$ mbpoll -m tcp -p 502 -a 1 -r 2 -v 127.0.0.1 42  #port 502, slave 1, start reference aka register 2, value 42, verbose, generated from localhost`

I get the error 

> mbpoll: Connection failed: Connection refused

I thought mbpoll could simulate a slave, because the output from it says "mbpoll 0.1-23 - FieldTalk(tm) Modbus(R) Master Simulator". Please tell me what command line parameters I should use. Thanks!

mmalyj, 2017-11-16:
I get it, mbpoll is a Modbus MASTER simulator. 

EpsilonRT, do you have a similar Modbus SLAVE Simulator that can be run from the command line? I need to specify TCP, and even if it simulates just one slave and a value for one register, that would be adequate.

epsilonrt, 2017-11-16:
Hi,
mbpoll is indeed a master. I do not know any slave "open source" but libmosbus also allows to design MODBUS servers (http://libmodbus.org/docs/v3.1.4/).
You have this project https://github.com/taka-wang/c-modbus-slave/blob/master/src/server.c that could allow you to simulate a simple slave.
I do not need this feature in my work, so for now, it is not planned to integrate with mbpoll.
regards.

closed 2017-11-16 by epsilonrt