epsilonrt/mbpoll #10: Modbus read RTU gives timeout

rjbaat, 2018-01-31:
Hi,
I have a problem with an embeddedpi with mbpoll reading modbus registers on ttyAMA0 from a simply modbus slave running on windows. The laptop has an rs485 to usb converter like on the picture. It has been working before but now i only get timeouts.

When i issue the command, i get this respons:
```
root@raspberrypi:~# mbpoll -b 9600 -m rtu -a 1 -r 1 -c 1 /dev/ttyAMA0 -1 -P none
mbpoll 0.1-23 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright (c) 2015 epsilonRT, All rights reserved.
This software is governed by the CeCILL license <http://www.cecill.info>

Protocol configuration: Modbus RTU
Slave configuration...: address = [1]
                        start reference = 1, count = 1
Communication.........: /dev/ttyAMA0,       9600-8N1 
                        t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register, output (holding) register table

-- Polling slave 1...
Read output (holding) register failed: Connection timed out
```

Is there something i can do to debug this?

[image]
[image]

epsilonrt, 2018-01-31:
Hello ,
The first probable cause is the half-duplex switching of the RS485 driver. If you can check with an oscilloscope that the DE signal is correctly applied. You do not say how the RS485 is handled on your embedded board. But normally you should have a -4 or -5 option in mbpoll. So I think the RS485 driver on this side is stuck in transmitter (DE = 1)

Small precision, without parity we must have 2 stop bits to remain the standard OSL modbus. In fact, the time between frames is 3.5 bytes min. and one byte is composed of 11 bits. As you have a virtual slave this can be the cause. RTU is a very restrictive protocol in real time.

You can help yourself with the mbpoll -v option to see what is received by mbpoll. The default timeout is 1 second, you can also extend it.

I think you should quickly find the root of the problem, thank you for bringing back the cause of the problem once found.
Best regards

rjbaat, 2018-01-31:
Well i believe there is hardware switching on the board. And because of it working before i don't think that would be the case tbh. 

Also i tried with even parity. But this has the same result. I see the TX and RX lights on the converter light up when doing a read and the software is showing a fine read and respons. But the box isn't getting the respons. 

Then with the -v i see this result:

```
root@raspberrypi:~# mbpoll -b 9600 -m rtu -a 1 -r 1 -c 1 /dev/ttyAMA0 -1 -P even -v -o 2
mbpoll 0.1-23 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright (c) 2015 epsilonRT, All rights reserved.
This software is governed by the CeCILL license <http://www.cecill.info>

Opening /dev/ttyAMA0 at 9600 bauds (E, 8, 1)
Protocol configuration: Modbus RTU
Slave configuration...: address = [1]
                        start reference = 1, count = 1
Communication.........: /dev/ttyAMA0,       9600-8E1 
                        t/o 2.00 s, poll rate 1000 ms
Data type.............: 16-bit register, output (holding) register table

-- Polling slave 1...
[01][03][00][00][00][01][84][0A]
Waiting for a confirmation...
ERROR Connection timed out: select
Read output (holding) register failed: Connection timed out
```

What i will try tomorrow is to use another laptop. And i will try to do a box <--> box connection instead of a box <--> rs485 usb converter <--> laptop connection.

epsilonrt, 2018-01-31:
If mbpoll does not get the answer it's because the linux driver of the serial port is not getting it. As your slave receives the request of the master and sends an answer is that it is lost on the way:
- either the embeddedpi RS485 driver is in transmitter mode
- the signals are below the threshold.
Warning ! you still have not put the -4 option to mbpoll!
I have 12 copies of USB-RS485 FTDI adapter and I took a week to run them. My solution was to put the bias resistors directly on the adapter. Since then, I have no problem.
I am interrogative about the sentence:
"_Isolated 2-Wire RS485 Interface with transparent hardware flow control and on-board bus pull and termination resistors._"
Only an analysis with an oscilloscope will allow you to remove the doubt.
Good luck.

rjbaat, 2018-02-01:
oke i tried the command with a new longer 2 wire cable. Since i thought it might be to short. Also added the -4 option. But the result is still the same :(.

```
mbpoll -a 1 -b 9600 -P none -t 4 -r 1 -c 5 /dev/ttyAMA0 -1 -v
mbpoll 0.1-23 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright (c) 2015 epsilonRT, All rights reserved.
This software is governed by the CeCILL license <http://www.cecill.info>

Opening /dev/ttyAMA0 at 9600 bauds (N, 8, 1)
Protocol configuration: Modbus RTU
Slave configuration...: address = [1]
                        start reference = 1, count = 5
Communication.........: /dev/ttyAMA0,       9600-8N1 
                        t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register, output (holding) register table

-- Polling slave 1...
[01][03][00][00][00][05][85][C9]
Waiting for a confirmation...
ERROR Connection timed out: select
Read output (holding) register failed: Connection timed out

```
Also on advise of the developer of embedded I did try to remove the 2 jumpers for lk5 and lk6. But that didnt make a difference aswell.

I am going to try to do a connection between 2 boxes. To see if that works.

epsilonrt, 2018-02-01:
you do not understand, it's not the -t 4 option but the -4 option, look at the help page.

rjbaat, 2018-02-01:
You are right. I do understand, but I didn't look right. Tried again: 

```
root@raspberrypi:~# mbpoll -a 1 -b 9600 -P even -t 4 -r 1 -c 1 /dev/ttyAMA0 -1 -v -4
mbpoll 0.1-23 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright (c) 2015 epsilonRT, All rights reserved.
This software is governed by the CeCILL license <http://www.cecill.info>

Opening /dev/ttyAMA0 at 9600 bauds (E, 8, 1)
Protocol configuration: Modbus RTU
Slave configuration...: address = [1]
                        start reference = 1, count = 1
Communication.........: /dev/ttyAMA0,       9600-8E1 
                        t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register, output (holding) register table

-- Polling slave 1...
[01][03][00][00][00][01][84][0A]
Waiting for a confirmation...
ERROR Connection timed out: select
Read output (holding) register failed: Connection timed out
```

rjbaat, 2018-02-01:
I have setup a minicom to minicom serial connection between the boxes and i can get the messages back and forth between the boxes. But when doing the same with an hypterminal -> minicom connection on the box i only get data from the box but can't receive it on the box.

Serial connection:
box -> windows (message received = OK)
box <- windows (message NOT received)
box -> box (message received = OK)
box <- box (message received = OKE)

So i think maybe the usb converters are broken somehow. I will have to get others to see if that is the problem. In the meantime i will try to get modbus slave software setup on one of the boxes. Do you maybe have a suggestion for this?

epsilonrt, 2018-02-01:
I understand that you do not have measuring equipment? you may not be an electronics engineer? For your USB adapter problem, I already had to change the MAX485 which is on it ... What is strange is that you get well, so for me I stay on my diagnosis, the signal DE MAX485 is stuck in the high state. Have you tried to change the flow control at the Windows level? (normally FTDI circuits do not need hardware flow control).

I have already been asked the question, no I have no proposal. I do not need this function because I work as an electronics engineer and I have modbus slaves (with ATMEL MCUs).

I will have to think about integrating this function in mbpoll. Indeed, libmodbus allows to create servers (slaves) modbus ... If you are interested you can open another ticket, if other people are, I will see according to my available time ...

rjbaat, 2018-02-02:
Correct I do not have measurement equipment since I am not an electronics engineer or have knowledge of this. On your second question on hyperterminal I have indeed disabled the hardware control on the connection settings for the test. Is that what you mean? Not sure if this would be possible in the modbus slave software to. 

Well i do not have hardware slaves atm. But I will get some from my client soon. But for testing and debugging it would indeed be helpful to be able to setup a slave simulator on linux besides the current versions there are for windows.

rjbaat, 2018-02-02:
I have tested with a different usb to rs485 converter from Sintechi now and it works again :). Thanks for the help!

root@raspberrypi:~# mbpoll -a 1 -b 9600 -P none -t 4 -r 1 -c 5 /dev/ttyAMA0 -1 -v -4
mbpoll 0.1-23 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright (c) 2015 epsilonRT, All rights reserved.
This software is governed by the CeCILL license <http://www.cecill.info>

Opening /dev/ttyAMA0 at 9600 bauds (N, 8, 1)
Protocol configuration: Modbus RTU
Slave configuration...: address = [1]
                        start reference = 1, count = 5
Communication.........: /dev/ttyAMA0,       9600-8N1 
                        t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register, output (holding) register table

-- Polling slave 1...
[01][03][00][00][00][05][85][C9]
Waiting for a confirmation...
<01><03><0A><00><00><00><00><00><00><00><00><00><00><24><B6>
[1]: 	0
[2]: 	0
[3]: 	0
[4]: 	0
[5]: 	0

epsilonrt, 2018-02-02:
Okay, so it was a hardware problem.

closed 2018-02-02 by epsilonrt