epsilonrt/mbpoll #12: indicate toolchain used to build on Windows

jonadem, 2018-03-01:
I tried to build mbpoll on Windows 10 without success so far.
I tried with MS Visual Studio 2013 and with MinGW but it always fails (with MinGW, I was able to build mbpoll.c but the linker still gives me errors).
An hint will be highly appreciated !

epsilonrt, 2018-03-01:
Hi
Did you read the INSTALL file? 
https://translate.googleusercontent.com/translate_c?depth=1&rurl=translate.google.com&sl=auto&sp=nmt4&tl=en&u=https://github.com/epsilonrt/mbpoll/blob/master/INSTALL&xid=17259,15700002,15700023,15700105,15700124,15700149,15700168,15700173,15700201&usg=ALkJrhjY5NMKB2TiQh5DCES0chhZvz5yWA
I do not have (and never would have ...) Windows 10 so I can not check. But to compile Mbpoll under windows you need a microsoft MSVC compiler free version 2013 works very well.

epsilonrt, 2018-03-01:
I added an setup file of an old version that works very well (version 0.2) if it can save you time in your job.

jonadem, 2018-03-01:
>  Did you read the INSTALL file?
Sorry, I went through the README a little bit too fast and I missed the reference to the INSTALL file.
Anyway, I did basically what is explained.

Long story short, getopt.{h,c} is missing in MSVS, I need to provide ones.
Last issue was that I needed to link with Ws2_32.lib (I will send you a patch).
I succeeded !

Thank you,

BTW, it builds without complain with debian stretch ! ;-)

epsilonrt, 2018-03-01:
Great !

closed 2018-03-01 by epsilonrt