epsilonrt/mbpoll #15: Does this program support function code?

terencechuen, 2018-03-02:
In my environment, need to use function code, would you like to support it?

epsilonrt, 2018-03-02:
This page is intended to report mbpoll bugs or to propose new features. your question seems to indicate that you have a very vague idea of how MODBUS works. I advise you to head to one of the multiple forums dedicated to MODBUS.

closed 2018-03-02 by epsilonrt

0xStuart, 2023-08-04:
Seems like a reasonable request.  mbpoll is sending out function code 3, and the slave I am reading is expecting 4.  There doesn't appear to be a way of changing it.

terencechuen, 2023-08-04:
@0xStuart You may try "-t 4" parameter, the default is 16-bit output (holding) register data type.

[image]