epsilonrt/mbpoll #16: OS X

niraviry, 2018-03-24:
Hi.

I tried compiling under OS X latest version.
The cake fails on the getgitversion line 9 in the Cmakechecklist.
I am not an expert but this error halts the make.
BTW, there is no Modbus cli for OS X that supports serial bus.

niraviry, 2018-03-27:
Hi again.

It looks like the original issue was caused from the fact that the GitVersion file was missing when downloaded directly and to from the Git app.
Now the cmake works but the make shows errors all under OS X. I am aware to the fact the OS X is not in the list but it will be nice to have it working.

epsilonrt, 2018-03-28:
Sorry for my late response.
I do not have a mac so I can not see the problem.
So you'd have to copy the error output.
Is git installed on your machine?
where did you get gitversion.cmake?

epsilonrt, 2018-03-28:
from what I understand you do not use GIT, so I modified gitversion; tell me if that solves your problem.
Another possibility is that cmake does not handle file paths properly so if it is not resolved, you have to send me the error output.

niraviry, 2018-03-28:
Hi.

Thanks for answering.

Yes I have git version 2.16.2 installed. I am not a git expert (I am a
hardware engineer). My understanding was that since the is a call to git
functions during Cmake I need it installed.
I have used your project as a download using a browser which eventually
missed an important file - the version-git.h file which caused the Cmake
compilation to stop a line 9 of the Cmakelists.txt file.
Once I have noticed that, I have cloned the repository using Git desktop
lacally and this caused the Cmake to end up with no errors.
Now I have errors running the make.
It reports syntax errors.
Tell me what data do you want as it seems like all version of Modbus I have
tried under OS X suffers from some errors or unchecked version under OS X.
This happens also with libmodbus versions that are defined as OS X
compatible.

niraviry, 2018-03-28:
Hi again.

Where did you modify the file in?

I also attach the make output. See at the top it identifies git.

Aviry-Ronits-Mac-mini:build MyServer$ make

-- Found Git: /usr/local/bin/git (found version "2.16.2")

-- Configuring done

-- Generating done

-- Build files have been written to:
/Users/MyServer/Documents/github/mbpoll/build

*Scanning dependencies of target mbpoll*

[ 12%] Building C object CMakeFiles/mbpoll.dir/src/mbpoll.c.o

*/Users/MyServer/Documents/github/mbpoll/src/mbpoll.c:1192:11:
**warning: **implicit
declaration of function 'basename' is*

*      invalid in C99 [-Wimplicit-function-declaration]*

          basename (progname), VERSION_SHORT);

*          ^*

*/Users/MyServer/Documents/github/mbpoll/src/mbpoll.c:1192:11:
**warning: **format
specifies type 'char *' but the argument*

*      has type 'int' [-Wformat]*

          basename (progname), VERSION_SHORT);

*          ^~~~~~~~~~~~~~~~~~~*

*/Users/MyServer/Documents/github/mbpoll/src/mbpoll.c:1201:10:
**warning: **incompatible
integer to pointer conversion*

*      initializing 'char *' with an expression of type 'int'
[-Wint-conversion]*

  char * sMyName =  basename (progname);

*         ^          ~~~~~~~~~~~~~~~~~~~*

3 warnings generated.

[ 12%] Building C object CMakeFiles/mbpoll.dir/3rdparty/modbus/modbus.c.o

In file included from
/Users/MyServer/Documents/github/mbpoll/3rdparty/modbus/modbus.c:24:

*/Users/MyServer/Documents/github/mbpoll/3rdparty/modbus/modbus-private.h:111:8:
**error: **expected parameter declarator*

size_t strlcpy(char *dest, const char *src, size_t dest_size);

*       ^*

*/usr/include/secure/_string.h:108:47: note: *expanded from macro 'strlcpy'

                __builtin___strlcpy_chk (dest, __VA_ARGS__, __darwin_obsz
(dest))
[... 251 lines cut ...]

*      '__builtin___strlcpy_chk'*

size_t strlcpy(char *dest, const char *src, size_t dest_size)

*       ^*

*/usr/include/secure/_string.h:108:3: note: *expanded from macro 'strlcpy'

                __builtin___strlcpy_chk (dest, __VA_ARGS__, __darwin_obsz
(dest))

*                ^*

2 warnings and 7 errors generated.

make[2]: *** [CMakeFiles/mbpoll.dir/3rdparty/modbus/modbus.c.o] Error 1

make[1]: *** [CMakeFiles/mbpoll.dir/all] Error 2

make: *** [all] Error 2

Thanks,

Nir

niraviry, 2018-03-29:
Hi.

To clarify things.
Your update changed nothing. I got the git file so your update was not
needed and errors are the same.

Hope you can solve this.

Nir

בתאריך יום ד׳, 28 במרץ 2018, 15:11, מאת epsilonrt ‏<notifications@github.com

epsilonrt, 2018-03-29:
Hi,
I did not understand that at all. I created a r1.4-osx branch to do the tests.
there is a problem with the basename () and strlcpy () functions.
Can you switch to this branch and send me the result of the compilation as an attached file?

niraviry, 2018-03-29:
OK.
Will do that when I get back to my PC.
Thanks.

בתאריך יום ה׳, 29 במרץ 2018, 11:15, מאת epsilonrt ‏<notifications@github.com

niraviry, 2018-03-29:
Hi.

Looks the same after using the new version.

See below:

-- Build files have been written to:
/Users/MyServer/Documents/Github/mbpoll/build

Aviry-Ronits-Mac-mini:build MyServer$ make

*Scanning dependencies of target mbpoll*

[ 12%] Building C object CMakeFiles/mbpoll.dir/src/mbpoll.c.o

[ 25%] Building C object CMakeFiles/mbpoll.dir/3rdparty/modbus/modbus.c.o

In file included from
/Users/MyServer/Documents/Github/mbpoll/3rdparty/modbus/modbus.c:24:

*/Users/MyServer/Documents/Github/mbpoll/3rdparty/modbus/modbus-private.h:111:8:
**error: **expected parameter declarator*

size_t strlcpy(char *dest, const char *src, size_t dest_size);

*       ^*

*/usr/include/secure/_string.h:108:47: note: *expanded from macro 'strlcpy'

                __builtin___strlcpy_chk (dest, __VA_ARGS__, __darwin_obsz
(dest))

*                                                            ^*

*/usr/include/secure/_common.h:39:62: note: *expanded from macro
'__darwin_obsz'

#define __darwin_obsz(object) __builtin_object_size (object,
_USE_FORTIFY_LEVEL > 1 ? 1 : 0)

*                                                             ^*

*/usr/include/secure/_common.h:30:32: note: *expanded from macro
'_USE_FORTIFY_LEVEL'

#    define _USE_FORTIFY_LEVEL 2

*                               ^*

In file included from
/Users/MyServer/Documents/Github/mbpoll/3rdparty/modbus/modbus.c:24:

*/Users/MyServer/Documents/Github/mbpoll/3rdparty/modbus/modbus-private.h:111:8:
**error: **expected ')'*

*/usr/include/secure/_string.h:108:47: note: *expanded from macro 'strlcpy'

                __builtin___strlcpy_chk (dest, __VA_ARGS__, __darwin_obsz
(dest))

*                                                            ^*

*/usr/include/secure/_common.h:39:62: note: *expanded from macro
'__darwin_obsz'

#define __darwin_obsz(object) __builtin_object_size (object,
_USE_FORTIFY_LEVEL > 1 ? 1 : 0)

*                                                             ^*

[... 208 lines cut ...]

size_t strlcpy(char *dest, const char *src, size_t dest_size)

*       ^*

*/usr/include/secure/_string.h:108:3: note: *expanded from macro 'strlcpy'

                __builtin___strlcpy_chk (dest, __VA_ARGS__, __darwin_obsz
(dest))

*                ^*

2 warnings and 7 errors generated.

make[2]: *** [CMakeFiles/mbpoll.dir/3rdparty/modbus/modbus.c.o] Error 1

make[1]: *** [CMakeFiles/mbpoll.dir/all] Error 2

make: *** [all] Error 2

Aviry-Ronits-Mac-mini:build MyServer$

Thanks.

Nir

epsilonrt, 2018-03-29:
no the error is different, the basename () function is still correctly set. However, the display is inconsistent. Can you send me the result of `git describe`?

niraviry, 2018-03-29:
> you should use git :-) here is a summary, launch the BASH console:
> cd
> git clone https://github.com/epsilonrt/mbpoll.git
> cd mbpoll
> git checkout r1.4-osx
> mkdir build
> cd build
> cmake ..
> make

OK.

I did most of it except for the checkout. Testing.

thanks.

niraviry, 2018-03-29:
Hi.

It looks the same:

-- Build files have been written to:
/Users/MyServer/Documents/Github/mbpoll/build

*Scanning dependencies of target mbpoll*

[ 12%] Building C object CMakeFiles/mbpoll.dir/src/mbpoll.c.o

[ 25%] Building C object CMakeFiles/mbpoll.dir/3rdparty/modbus/modbus.c.o

In file included from
/Users/MyServer/Documents/Github/mbpoll/3rdparty/modbus/modbus.c:24:

*/Users/MyServer/Documents/Github/mbpoll/3rdparty/modbus/modbus-private.h:111:8:
**error: **expected parameter declarator*

size_t strlcpy(char *dest, const char *src, size_t dest_size);

*       ^*

*/usr/include/secure/_string.h:108:47: note: *expanded from macro 'strlcpy'

                __builtin___strlcpy_chk (dest, __VA_ARGS__, __darwin_obsz
(dest))

*                                                            ^*

*/usr/include/secure/_common.h:39:62: note: *expanded from macro
'__darwin_obsz'

#define __darwin_obsz(object) __builtin_object_size (object,
_USE_FORTIFY_LEVEL > 1 ? 1 : 0)

*                                                             ^*

*/usr/include/secure/_common.h:30:32: note: *expanded from macro
'_USE_FORTIFY_LEVEL'

#    define _USE_FORTIFY_LEVEL 2

*                               ^*

In file included from
/Users/MyServer/Documents/Github/mbpoll/3rdparty/modbus/modbus.c:24:

*/Users/MyServer/Documents/Github/mbpoll/3rdparty/modbus/modbus-private.h:111:8:
**error: **expected ')'*

*/usr/include/secure/_string.h:108:47: note: *expanded from macro 'strlcpy'

                __builtin___strlcpy_chk (dest, __VA_ARGS__, __darwin_obsz
(dest))

*                                                            ^*

*/usr/include/secure/_common.h:39:62: note: *expanded from macro
'__darwin_obsz'

#define __darwin_obsz(object) __builtin_object_size (object,
_USE_FORTIFY_LEVEL > 1 ? 1 : 0)

*                                                             ^*

*/usr/include/secure/_common.h:30:32: note: *expanded from macro
'_USE_FORTIFY_LEVEL'

#    define _USE_FORTIFY_LEVEL 2
[... 213 lines cut ...]

*                ^*

2 warnings and 7 errors generated.

make[2]: *** [CMakeFiles/mbpoll.dir/3rdparty/modbus/modbus.c.o] Error 1

make[1]: *** [CMakeFiles/mbpoll.dir/all] Error 2

make: *** [all] Error 2

Aviry-Ronits-Mac-mini:Github MyServer$

The git describe now (in the build folder) is:

Aviry-Ronits-Mac-mini:build MyServer$ git describe

v1.3-4-g444b009

I think I have run that in the same folder in the 1st place so here there
is a change.

Thans.

2018-03-29 21:28 GMT+03:00 Nir Aviry <aviryn@gmail.com>:

epsilonrt, 2018-03-29:
the r1.4-osx branch has been modified, can you redo a compilation after `git pull` ?

niraviry, 2018-03-30:
Hi.

Almost done. Now just 1 error:

*Scanning dependencies of target mbpoll*

[ 12%] Building C object
CMakeFiles/mbpoll.dir/3rdparty/modbus/modbus-rtu.c.o

*/Users/MyServer/Documents/Github/mbpoll/3rdparty/modbus/modbus-rtu.c:27:10:
**fatal error: **'linux/serial.h' file not found*

#include <linux/serial.h>

*         ^~~~~~~~~~~~~~~~*

1 error generated.

make[2]: *** [CMakeFiles/mbpoll.dir/3rdparty/modbus/modbus-rtu.c.o] Error 1

make[1]: *** [CMakeFiles/mbpoll.dir/all] Error 2

make: *** [all] Error 2

Waiting for your update.

Thanks.

niraviry, 2018-03-30:
Hi.

Looked at it again.
It seems like you are directing the make to the wrong location in 3rd party
folder.

Thanks.

epsilonrt, 2018-03-30:
the error has been solved. I'm going to separate the source code of libmodbus as soon as I could integrate my change proposals to stephane.

niraviry, 2018-03-30:
OK.
Great.
I also have another proposal. Enable 32 bit reading. All metered devices
use 32 bit as it is what is needed for voltage and current.
I will check it out later on.

Thank.

בתאריך יום ו׳, 30 במרץ 2018, 10:42, מאת epsilonrt ‏<notifications@github.com

epsilonrt, 2018-03-30:
I understand that everything works?
if you read the help of mbpoll you will see that this function is already implemented. for t3 and t4 you have the suffixes :int and :float that allow you to read sensors. The -B option allows to choose the order of bytes.

niraviry, 2018-03-30:
Hi.

I did not say it is working. I said I will check.
There is another include that has the wrong path (and may be more as the
compilation stops in the middle):

Aviry-Ronits-Mac-mini:build MyServer$ make

*Scanning dependencies of target mbpoll*

[ 12%] Building C object
CMakeFiles/mbpoll.dir/3rdparty/modbus/modbus-rtu.c.o

[ 25%] Building C object CMakeFiles/mbpoll.dir/src/mbpoll.c.o

[ 37%] Building C object CMakeFiles/mbpoll.dir/3rdparty/modbus/modbus.c.o

[ 50%] Building C object
CMakeFiles/mbpoll.dir/3rdparty/modbus/modbus-data.c.o

*/Users/MyServer/Documents/Github/mbpoll/3rdparty/modbus/modbus-data.c:29:12:
**fatal error: **'byteswap.h' file not found*

#  include <byteswap.h>

*           ^~~~~~~~~~~~*

1 error generated.

make[2]: *** [CMakeFiles/mbpoll.dir/3rdparty/modbus/modbus-data.c.o] Error 1

make[1]: *** [CMakeFiles/mbpoll.dir/all] Error 2

make: *** [all] Error 2

I think it is better to check all include statements.

Thanks.

niraviry, 2018-03-30:
Hi again.

Had few minuts to check it out myself. OS X has no bytswap nor endian.
I crossed a solution for byteswap as follows:
https://gist.github.com/atr000/249599

I am sure the endian has one too.

Thanks,

NIr

niraviry, 2018-03-31:
Hi again.
Had time to check things and I have fixed the error as the if condition of the byteswap was wrong.
Anyway compilation ends up with no errors but many flags are not working like -b -P -s etc.

Some do work.

niraviry, 2018-04-01:
Hi. From some reason I fot not see my enhanced replay.
Almost all the parameters are resonded with illigal parameter error.
I have managed to firce the parameters to suit my own combination and it
works including big /little endian but still parameters are not passed so I
cant control the address I need to read.
I also had to remove the include byteswapcondition to make it end the make
compilation.
Thank a lot.

בתאריך יום ו׳, 30 במרץ 2018, 21:58, מאת Nir Aviry ‏<aviryn@gmail.com>:

closed 2018-09-20 by epsilonrt