epsilonrt/mbpoll #22: Can't build with CentOS 7

DanielCuneo, 2018-10-11:
I'm trying to build your tools on CentOS 7. I started by installed the libmodbus in yum, but realized it was too old.
I then cloned  v1.4 from your repo and built that.
 
Cmake wasn't finding libmodbus at all, so I hacked the CMakeLists.txt by commenting out the find package part and just put in `/usr/local/lib` for the `link_dirs`.

cmake then appeared to run correctly, but unsure.
Followed up with `make` and getting errors:
`>make
Linking C executable mbpoll
CMakeFiles/mbpoll.dir/src/mbpoll.c.o: In function `vSigIntHandler':
mbpoll.c:(.text+0x35): undefined reference to `modbus_close'
mbpoll.c:(.text+0x41): undefined reference to `modbus_free'
CMakeFiles/mbpoll.dir/src/mbpoll.c.o: In function `vReportSlaveID':
mbpoll.c:(.text+0x31e): undefined reference to `modbus_set_slave'
mbpoll.c:(.text+0x363): undefined reference to `modbus_report_slave_id'
mbpoll.c:(.text+0x45e): undefined reference to `modbus_strerror'
CMakeFiles/mbpoll.dir/src/mbpoll.c.o: In function `main':
mbpoll.c:(.text.startup+0x5e0): undefined reference to `modbus_new_tcp_pi'
mbpoll.c:(.text.startup+0x603): undefined reference to `modbus_set_debug'
mbpoll.c:(.text.startup+0x63b): undefined reference to `modbus_connect'
mbpoll.c:(.text.startup+0x697): undefined reference to `modbus_set_response_timeout'
mbpoll.c:(.text.startup+0x750): undefined reference to `modbus_read_registers'
mbpoll.c:(.text.startup+0x76f): undefined reference to `modbus_strerror'
mbpoll.c:(.text.startup+0x7e6): undefined reference to `modbus_set_slave'
mbpoll.c:(.text.startup+0x846): undefined reference to `modbus_read_bits'
mbpoll.c:(.text.startup+0x899): undefined reference to `modbus_read_input_bits'
mbpoll.c:(.text.startup+0x8b9): undefined reference to `modbus_read_input_registers'
mbpoll.c:(.text.startup+0x8d6): undefined reference to `modbus_set_slave'
mbpoll.c:(.text.startup+0x911): undefined reference to `modbus_write_registers'
mbpoll.c:(.text.startup+0x930): undefined reference to `modbus_strerror'
mbpoll.c:(.text.startup+0xa96): undefined reference to `modbus_write_bits'
mbpoll.c:(.text.startup+0xaeb): undefined reference to `modbus_new_rtu'
mbpoll.c:(.text.startup+0xb59): undefined reference to `modbus_free'
mbpoll.c:(.text.startup+0xb65): undefined reference to `modbus_strerror'
mbpoll.c:(.text.startup+0xc8c): undefined reference to `modbus_write_bit'
mbpoll.c:(.text.startup+0xcad): undefined reference to `modbus_write_register'
mbpoll.c:(.text.startup+0xdc7): undefined reference to `modbus_rtu_set_serial_mode'
mbpoll.c:(.text.startup+0xdd9): undefined reference to `modbus_rtu_set_rts'
collect2: error: ld returned 1 exit status
make[2]: *** [mbpoll] Error 1
make[1]: *** [CMakeFiles/mbpoll.dir/all] Error 2
make: *** [all] Error 2
` 

This seems like the `modbus.h` header is missing but it's in `/usr/local/include/modbus`.

Any help is appreciated, I think your tools will help me avoid Windows apps.
TIA

epsilonrt, 2018-10-12:
Hello,
libmodbus is not installed correctly, you must follow the procedure: https://github.com/epsilonrt/mbpoll#installation. If you install libmodbus properly, you do not need to modify anything. Once installed libmodbus, delete everything in the cmake build directory.
Do not forget the ldconfig!

greearb, 2019-09-17:
I am trying to compile on Fedora-29.  It too has a too-old libmodbus.
I cloned latest libmodbus, built, and ran 'make install'.  Files show up in /usr/local:

/usr/local/lib/libmodbus.la
/usr/local/lib/libmodbus.so
/usr/local/lib/libmodbus.so.5
/usr/local/lib/libmodbus.so.5.1.0
/usr/local/lib/pkgconfig/libmodbus.pc
/usr/local/include/modbus/modbus.h

But, cmake cannot find it:

[greearb@ben-dt4 build]$ cmake ../
-- Checking for module 'libmodbus>=3.1.4'
--   Package 'libmodbus', required by 'virtual:world', not found
CMake Error at /usr/share/cmake/Modules/FindPkgConfig.cmake:457 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake/Modules/FindPkgConfig.cmake:642 (_pkg_check_modules_internal)
  CMakeLists.txt:42 (pkg_check_modules)

greearb, 2019-09-17:
This makes it compile for me:

diff --git a/CMakeLists.txt b/CMakeLists.txt
index 8e10f4a..6b69bc5 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -39,7 +39,11 @@ endif (NOT CL_USED)
 set(CMAKE_MODULE_PATH "${CMAKE_CURRENT_SOURCE_DIR}/cmake" ${CMAKE_MODULE_PATH})
 
 find_package(PkgConfig REQUIRED)
-pkg_check_modules (LIBMODBUS REQUIRED libmodbus>=3.1.4)
+#pkg_check_modules (LIBMODBUS REQUIRED libmodbus>=3.1.4)
+
+set(LIBMODBUS_INCLUDE_DIRS /usr/local/include/modbus)
+set(LIBMODBUS_LIBRARY_DIRS /usr/local/lib)
+set(LIBMODBUS_LIBRARIES modbus)
 
 link_directories(${LIBMODBUS_LIBRARY_DIRS})
 add_definitions(${LIBMODBUS_CFLAGS})

epsilonrt, 2019-09-18:
Hello,
While your solution allows you to compile but the problem is not solved.
You must have libmodbus version greater than or equal to 3.14.
You tell me that you have compiled and installed libmodbus but what is its version? can you find it in the version.h file that is in libmodbus sources?

greearb, 2019-09-18:
I don't see a version.h file, but here is the git commit ID:

ommit f1eb4bc7ccb09cd8d19ab641ee37637f8c34d16d (HEAD -> master, origin/master, origin/HEAD)
Author: i-ky <gl.ivanovsky@gmail.com>
Date:   Tue Jul 10 15:58:45 2018 +0300

     Fixed MODBUS_GET_* macros in case of negative values

     In case resulting value should be negative it is incorrect to use '+' operator to construct it from pieces, because highest bytes will result in negative number after bitwise shift while others will stay positive. Replacing addition with '|' should solve the issue.

 From repository:

url = https://github.com/stephane/libmodbus.git

Thanks,
Ben

cristianlivella, 2022-04-06:
In case it will help anyone, I was having the same problem with Fedora 33, and I solved it by running this before `sudo ldconfig`.

```
echo "/usr/local/lib" | sudo tee -a /etc/ld.so.conf
```

nightcatnl, 2023-06-07:
i downloaded libmodmus (3.1.10) from repo.
Compiled it
Installed it.
test scripts (server and client) work fine.

Though I am not able to compile the mbpoll because of the above error

Why can cmake not find it ?

Really don't get it.