epsilonrt/mbpoll #30: address greater then 255

nicolasbussieres, 2019-12-05:
Hello, im trying to read modbus from a cisco , specifing addresses greater then 255
https://www.cisco.com/c/en/us/td/docs/switches/connectedgrid/MODBUS/swmodreg_ie2k.html#pgfId-1092936

but when i try to use mbpoll, i get the error :
# mbpoll  -a 800 -r 64  x.x.x.x
mbpoll: slave address out of range (800) ! Try -h for help.

epsilonrt, 2019-12-05:
Hi,
It seems very problematic to code 800 on 1 byte, but the slave address is coded on a byte!
Please consult the MODBUS OSL and TCP specifications before ...
http://www.modbus.org/specs.php

closed 2019-12-05 by epsilonrt

nicolasbussieres, 2019-12-05:
thanks, figured out the probleme. the addresse is the register ....
for future reference, the following allows polling on cisco devices (IE switches)

for product ID on cisco docs:

Address | Number of Registers | Description | R/W | Format | Example/Note
800 | 64 | Product ID | R | Text | “IE-2000-16TC-G-E”

_convert 800(hex) to  2048
count 64
First reference is 0
type is hex_  
 mbpoll -r 2048 -c64 -1 -0 -t 4:hex  x.x.x.x

output :
-- Polling slave 1...
[2048]:      0x4945
[2049]:      0x2D32
[2050]:      0x3030
[2051]:      0x302D
[2052]:      0x3136
[2053]:      0x5443
[2054]:      0x2D47
[2055]:      0x2D45
[2056]:      0x0000
...

hex to ascii : IE-2000-16TC-G-E