epsilonrt/mbpoll #38: Missing parameter on Windows

epsilonrt, 2020-05-14:
But when I run it on windows 2016 (same command/parameters)
(It work well on linux)
 mbpoll -c 10  -B -t 4:float -1 192.168.0.100
mbpoll always show the error like this: 
[image]

_Originally posted by @johnwmail in https://github.com/epsilonrt/mbpoll/issues/37#issuecomment-628691245_

johnwmail, 2020-05-15:
Hi epsilonrt, thanks for your help.
But after added -m tcp -p 502, still not work on windows

[image]

Any idea, how to fix it?
Thanks.

johnwmail, 2020-05-19:
Hi, only me/windows 2016 have this problem?

epsilonrt, 2020-05-19:
Yes.
mbpoll 1 will no longer be updated, I will switch to mbpoll 2 with the integration of [libmodbuspp](https://github.com/epsilonrt/libmodbuspp).
I recommend that you recompile under Windows 16 from sources. You have to use cmake and the MS compiler.
Good luck.

closed 2020-05-19 by epsilonrt