epsilonrt/mbpoll #40: I cannot install using the quickstart guide for Raspbian

MissingCatPE, 2020-08-27:
I am still a newbie, but I have run this installation before using the quickstart guide and it worked fine.

I cannot seem to start the installation using the (this is under the raspbian section):

wget -O- http://www.piduino.org/piduino-key.asc | sudo apt-key add -
echo 'deb http://raspbian.piduino.org stretch piduino' | sudo tee /etc/apt/sources.list.d/piduino.list
sudo apt update
sudo apt install mbpoll

I get "HTTP request sent, awaiting response....404 Not Found."

The installation then terminates with the remaining steps having no data to refer to.

Am I doing something wrong? Please assist.

Regards,

MissingCatPE, 2020-08-28:
Still very new to all of this. If you are like me...this is how I got it to work, should the quickstart method not work for you.

git clone https://github.com/epsilonrt/mbpoll.git
sudo apt-get install cmake pkg-config
cd mbpoll
mkdir build
cd build
cmake ~/mbpoll/
make
sudo make install
sudo ldconfig

skupfer, 2020-09-03:
Since the key is not available currently, you would have to do it like this (also you tried to add stretch but have buster)
```
sudo echo 'deb [trusted=yes] http://apt.piduino.org buster piduino' | sudo tee -a /etc/apt/sources.list.d/mbpoll.list
sudo apt update
sudo apt install mbpoll
``` 

You'll get a warning for the missing key but can simply ignore it. 

REMINDER: Never ever do `[trusted=yes]` if you are not sure about the repository!

closed 2021-01-13 by MissingCatPE