epsilonrt/mbpoll #45: mbpoll giving a 'std::system_error' (Pi 4B 2G - Buster)

MissingCatPE, 2021-01-13:
Cannot seem to get mbpoll to run on Pi 4B 2GB running Buster.

terminate called after throwing an instance of 'std::system_error'
  what():  It seems that this system is not supported !: Operation not supported
Aborted

First time seeing the error. Replicated the error of a fresh install of Raspbian

Please assist, not sure if I have missed something.

victoriv, 2021-03-03:
I got the same error when following the quickstart guide for installing on buster.

However, I got mbpoll working when building from source :)

romor, 2021-07-31:
Same happening for me on a Pi 3 CM.
Also, building from source solved this issue.

JetForMe, 2021-10-09:
Oh, this seems to be related to piduino. `pinfo` returns this error.

ste48, 2022-10-29:
Hello,
same error with me. Any idea how to solve it?

iz8mbw, 2024-04-02:
On Raspberry Pi 3 I have had the same error
```
mbpool what():  It seems that this system is not supported !: Operation not supported
```
after executing "rpi-update" (as we know "rpi-update" upgrades Raspberry Pi Kernel).
In order to recover I execured:
```
sudo apt-get update
sudo apt-get install --reinstall raspberrypi-bootloader raspberrypi-kernel
```
and so I roll back to a more "stable" Kernel version and `mbpoll` started again to work.

This the current workig Kernel I'm using is `Linux rpi3 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`

iz8mbw, 2024-04-02:
ping @epsilonrt 
Thanks!