epsilonrt/mbpoll #51: start reference is off by -1

randy-, 2021-11-13:
Hello,

Testing mbpoll with Victron Energry Color GX. The Victron Energry Color GX error logs shows 
mbpoll 1.0-0 (Debian 1.4.11+dfsg-2~bpo10+1) starting reference (-r) is off by -1.
Example: ./mbpoll -1 -a 100 -t 3 -r 840 -c 1 192.168.10.180
mbpoll will starts at 839 not 840.

Fix source file mbpoll.c
change line 952~
iStartReg = ctx.piStartRef[j] - ctx.iPduOffset;
to
iStartReg = ctx.piStartRef[j];

splatch, 2022-08-02:
Modbus spec in early days was not clear on that hence `-1` and `+1` errors are happening. Its all about how end device index registers - if it does it from 0 or from 1, then its matter of client if he assumes any offsets. Its a feature of a protocol.
Switching it might help you with that device but break dozen of others.

syadnom, 2023-05-26:
splatch, thank you sir.  I didn't come here for that specific question but I couldn't get any data out of a mikrotik knot modbus tcp bridge to a mean well DRS-480-24 unit.  I didn't know the +/- 1 issue existed.

For anyone else that ends up searching and finding this, do hex to decimal and add 1 for the register address.

(modbus mikrotik knot meanwell mean well DRS)

the78mole, 2023-06-02:
I also see the -1 offset. If you enable debugging output (option `-v`), you can see, that it requests a different register address than the user requests on the command line. I think that is more confusing than it helps.

[image]

git-developer, 2023-06-19:
I believe this behavior is defined by the Modbus specification (didn't look it up, though). mbpoll has command line argument `-0` to shift the reference adresses from 1-based to 0-based.

cowo78, 2024-11-15:
IMHO it's worth making this situation clearer in the help message: -r is the OFFSET from register 1 (or register 0 if -0 is specified)