epsilonrt/mbpoll #59: Ubuntu 20.04 (Mac Parallel VM Arm64)

samuelting, 2022-04-18:
Hello there,

I recently try to install this program in Ubuntu 20.04 Arm64 using Parallel virtual machine. Errors are encountered when using APT repository from [piduino.org](http://apt.piduino.org/).

```
sudo add-apt-repository 'deb http://apt.piduino.org stretch piduino'
Hit:1 http://apt.piduino.org buster InRelease                                     
Get:2 http://apt.piduino.org stretch InRelease [2,836 B]                          
Hit:3 http://ports.ubuntu.com/ubuntu-ports focal InRelease
Get:4 http://ports.ubuntu.com/ubuntu-ports focal-updates InRelease [114 kB]
Get:5 http://ports.ubuntu.com/ubuntu-ports focal-backports InRelease [108 kB]                          
Get:6 http://ports.ubuntu.com/ubuntu-ports focal-security InRelease [114 kB]                           
Get:7 http://ports.ubuntu.com/ubuntu-ports focal-updates/main arm64 DEP-11 Metadata [278 kB]           
Get:8 http://ports.ubuntu.com/ubuntu-ports focal-updates/universe arm64 DEP-11 Metadata [389 kB]       
Get:9 http://ports.ubuntu.com/ubuntu-ports focal-backports/main arm64 DEP-11 Metadata [8,000 B]        
Get:10 http://ports.ubuntu.com/ubuntu-ports focal-backports/universe arm64 DEP-11 Metadata [30.8 kB]   
Get:11 http://ports.ubuntu.com/ubuntu-ports focal-security/main arm64 DEP-11 Metadata [40.6 kB]        
Get:12 http://ports.ubuntu.com/ubuntu-ports focal-security/universe arm64 DEP-11 Metadata [66.2 kB]    
Fetched 1,152 kB in 8s (148 kB/s)                                                                      
Reading package lists... Done
N: Skipping acquire of configured file 'piduino/binary-arm64/Packages' as repository 'http://apt.piduino.org stretch InRelease' doesn't support architecture 'arm64'
```

It indicates the distribution apt-repository does not support arm64. Can any body help to resolve this??

Thank you.

remmercier, 2022-04-19:
Same issue here when trying to install on ubuntu 20.04 server (working few months ago).
When we checked on apt.piduino.org repository, there is now only armhf architecture that is published. 
Is there any reason @epsilonrt why this project is only build a binary for armhf architecture and not for the others (i386, amd64,...)?

samuelting, 2022-05-03:
@remmercier after hours of troubleshooting, I worked around this issue by compiling this repo from source.

git-developer, 2023-06-19:
It seems that the repo now supports `amd64`, `arm64` and `armhf` for Ubuntu 20.04 (since 2023-06-08). Would you mind trying and close this issue respectively?

samuelting, 2023-06-20:
Hi @git-developer ,

Thanks for update. I will try it when I am available.

samuelting, 2023-06-20:
Hi @git-developer ,

I have just tried. It works for Ubuntu 22.04 Mac Parallel VM Arm64.

closed 2023-06-20 by samuelting