epsilonrt/mbpoll #75: I got an error using mbpoll:

AlfonsoAlejandre, 2024-01-09:
I pluged a waveshare usb->RS485 dongle and the port appeared is /dev/ttyACM0. I used the comand:
sudo mbpoll -a 255 -b 9600 -d 8 -s 1 -P none -t 3 -r 1 -c 2 /dev/ttyACM0
And I got:
mbpoll: modbus-rtu.c:110: _modbus_rtu_build_request_basis: Assertion `ctx->slave != -1' failed.
May you help me?

epsilonrt, 2024-01-10:
I advise you to consult https://www.modbus.org/docs/Modbus_over_serial_line_V1_02.pdf
Address 255 is illegal and in RTU, the number of bits must be 11, so without parity, 2 stops are required.

Malvineous, 2024-01-26:
In that document, on page 7, it says devices can only have unique addresses assigned from 1 to 247.

But then on page 8 it goes on to say addresses from 0 to 255 are valid, with 0 being a broadcast address and 248 to 255 being reserved for future use.

mbpoll does not currently allow submissions on this broadcast address 0 when in RTU mode, and it also does not permit using any of these reserved addresses even though the spec does not prevent their use, it just says you must not assign these IDs to devices.

There are a number of devices around that use address 0 for setting their assigned address (so you can change the device address when you don't know what it currently is set to) as well as devices that use 255 for a similar purpose, even though technically listening on address 255 is not defined in the standard.

I think mbpoll should at least allow you to send messages on all 255 addresses listed in the spec, because it's a very useful tool when debugging and being able to push the limits of the standard when you need to is what makes tools like this so powerful.

At the very least I think it should support address 0 out of the box given that this is defined in the standard.

Malvineous, 2024-04-09:
Also this seems like a similar issue to #39