epsilonrt/mbpoll #91: Failed to build mbpoll on Rocky Linux 9.4

snowuyl, 2024-11-19:
git clone https://github.com/stephane/libmodbus.git
cd libmodbus
./autogen.sh
./configure
make
sudo make install
cd ..
https://github.com/epsilonrt/mbpoll
cd mbpoll
mkdir build
cd build
cmake ..
Last metadata expiration check: 4:40:42 ago on Tue 19 Nov 2024 08:48:53 AM CST.
No match for argument: libmodbus-dev
Error: Unable to find a match: libmodbus-dev
mkdir: cannot create directory ‘build’: File exists
-- Checking for module 'libmodbus>=3.1.4'
--   Package 'libmodbus', required by 'virtual:world', not found
CMake Error at /usr/share/cmake/Modules/FindPkgConfig.cmake:607 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake/Modules/FindPkgConfig.cmake:829 (_pkg_check_modules_internal)
  CMakeLists.txt:61 (pkg_check_modules)

-- Configuring incomplete, errors occurred!

r00t-, 2025-01-08:
your distribution (i have never heard of "Rocky Linux") seems to contain some (broken) logic that tries to automatically install dependencies.
it generates an invalid package name, and fails.
you should report this at your distribution.
(this output is obviously from the `apk` package manager, executing `apk add libmodbus-dev`, which would not normally happen during a build.)
```
Last metadata expiration check: 4:40:42 ago on Tue 19 Nov 2024 08:48:53 AM CST.
No match for argument: libmodbus-dev
Error: Unable to find a match: libmodbus-dev
mkdir: cannot create directory ‘build’: File exists
-- Checking for module 'libmodbus>=3.1.4'
-- Package 'libmodbus', required by 'virtual:world', not found
CMake Error at /usr/share/cmake/Modules/FindPkgConfig.cmake:607 (message):
A required package was not found
Call Stack (most recent call first):
/usr/share/cmake/Modules/FindPkgConfig.cmake:829 (_pkg_check_modules_internal)
```

r00t-, 2025-01-08:
as to why it's not finding the libmodbus you installed,
you gave only very little details.
`make install` for libmodus should have installed  `libmodbus.pc` to `/usr/lib/pkgconfig/` or similar.
you should check if that file is there,
and check if `pkg-config --cflags libmodbus` works.